DroidAITech/ROS-Academy-for-Beginners #3: robot_spawn.launch file 错误

hisrg, 2018-04-06:
您好！老师，最近我在跟进你在MOOC上的课程，现在已经能跑robot_sim_demo中的范例了，但是有时候能加载gazebo 材质，有时候不能，就会报[robot_spawn.launch] is not a launch file 错误，当我进入robot_sim_demo 文件夹后，运行roslaunch robot_spawn.launch 命令时，会报Invalid<arg>tag:robo_sim_demo之类的错误，尤其是我重启Ubuntu之后。

sychaichangkun, 2018-04-06:
[robot_spawn.launch] is not a launch file  的错误一般是没有source setup.bash造成的
你可以把错误的截图发一下

hisrg, 2018-04-06:
能否提供您的Email地址？GitHub好像commit不能带图片。谢谢！

hisrg, 2018-04-06:
MOOC网上我评论了，有截图，用户名是hisrg

sychaichangkun, 2018-04-06:
好的 已经回复你了

closed 2018-04-06 by sychaichangkun

cucohhh, 2018-10-15:
tianhao@tianhao-VirtualBox:~$ roslaunch robot_sim_demp robot_spawn.launch
[robot_spawn.launch] is neither a launch file in package [robot_sim_demp] nor is [robot_sim_demp] a launch file name
The traceback for the exception was written to the log file
老师请问这个报错是问什么啊？
是我用虚拟机的问题吗？