DroidAITech/ROS-Academy-for-Beginners #6: 运行第二条，安装教学所需的依赖包报错

likui1123, 2018-06-25:
rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
slam_sim_demo: No definition of [hector_mapping] for OS [debian]

编译报错
 ROS-Academy-for-Beginners/slam_sim_demo/CMakeLists.txt:3 (find_package)

-- Could not find the required component 'slam_karto'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "slam_karto" with
  any of the following names:

    slam_kartoConfig.cmake
    slam_karto-config.cmake

  Add the installation prefix of "slam_karto" to CMAKE_PREFIX_PATH or set
  "slam_karto_DIR" to a directory containing one of the above files.  If
  "slam_karto" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/slam_sim_demo/CMakeLists.txt:3 (find_package)

lazyparser, 2018-06-27:
> Could not find the required component 'slam_karto'.

这个依赖关系安装了么？

lazyparser, 2018-07-14:
关闭没有后续反馈的 issues

closed 2018-07-14 by lazyparser

LeehanLee, 2023-06-28:
用的ubuntu18.04，尝试catkin_make也报这个错误，安装以下两个包后成功编译：
sudo apt-get install ros-melodic-slam-karto -y
sudo apt-get install ros-melodic-hector-mapping -y
但是，场景模型出不来。。。
[image]