DroidAITech/ROS-Academy-for-Beginners #10: 编译demo时报错

997931058, 2018-07-06:
老师您好，我按照MOOC课程操作的时候，操作到编译github clone的代码的时候出现如下错误。之前的操作都和视频上您的一样，希望老师解答。

CMakeFiles/param_demo.dir/src/param_demo.cpp.o: In function `main':
param_demo.cpp:(.text+0x70): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
param_demo.cpp:(.text+0xcc): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
param_demo.cpp:(.text+0x12d): undefined reference to `ros::param::get(std::string const&, int&)'
param_demo.cpp:(.text+0x19b): undefined reference to `ros::NodeHandle::getParam(std::string const&, int&) const'
param_demo.cpp:(.text+0x2ae): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x3c8): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x4e2): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x5fc): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x6e0): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
param_demo.cpp:(.text+0x774): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x88e): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0x97a): undefined reference to `ros::param::set(std::string const&, int)'
param_demo.cpp:(.text+0x9eb): undefined reference to `ros::NodeHandle::setParam(std::string const&, int) const'
param_demo.cpp:(.text+0xa75): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0xb85): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
param_demo.cpp:(.text+0xc6c): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'

lazyparser, 2018-07-14:
Hi @likui1123 可以去 https://q.droid.ac.cn 去提问和讨论这个问题。我们已经转到Q站了

lazyparser, 2018-07-14:
` AttributeError: 'module' object has no attribute 'Interpreter'`

错误的原因看起来在这里，可以从这个角度来搜索一下。

closed 2018-07-14 by lazyparser