DroidAITech/ROS-Academy-for-Beginners #12: can not rosrun the file robot_keyboard_teleop.py

jimsonbuilder, 2018-08-08:
can not rosrun the file robot_keyboard_teleop.py,when i do "rosrun robot_sim_demo robot_keyboard_teleop.py",it show the error info below:
Couldn't find executable named robot_keyboard_teleop.py below /home/jimson/catkin_ws/src/ROS-Academy-for-Beginners/robot_sim_demo
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/jimson/catkin_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/scripts/robot_keyboard_teleop.py
but i source the environment.
What should i do?

jimsonbuilder, 2018-08-08:
my email:870255589@qq.com,
hope someone can show me solution this ros question,thanks a lot.

nbsh2, 2020-08-20:
遇到了同样的问题。因为git-clone网速太慢，还一直失败。直接下载的ZIP解压的。想键盘控制遇到了同样的问题。我来尝试下能不能搞好。