DroidAITech/ROS-Academy-for-Beginners #13: 用melodic版本ros运行报错

1989Ryan, 2018-08-08:
用melodic版本的ros运行rosdep安装配置会报错, 很多包无法正常下载

jimsonbuilder, 2018-08-08:
应该会有缺少依赖包等error提示，按error提示，找到相应的依赖包安装就可以解决。
你可能用到的命令：sudo apt-cache search 你的缺省依赖包名*；
或者直接用apt-get install查找，比如你少了一个叫“gazebo-ros-control”的依赖，就可以这样：sudo apt-get install gazebo-ros-，然后按tab键两下，就会列出可选的包，如果给出的字符越少，列出的包会越多。重复上面的步骤直到所有依赖都安装完成，之后再rosdep init和rosdep update等操作。
如果不行更换国内的软件镜像源，主要是网易阿里云中科大清华软件源镜像。换源sudo apt-get update一下，再rosdep，如有错误再按上面的步骤进行操作。
依然不行，考虑更换ubuntu版本和ros版本，两者的官方推荐匹配版本已经给出，比如indigo对应14.04，kinectic对应16.04等等。

KennethYangle, 2018-08-10:
我的也是 错误如下
`kenneth@kenneth-lenovo-g50-75m:~/catkin_ws$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
robot_sim_demo: Cannot locate rosdep definition for [yocs_cmd_vel_mux]
navigation_sim_demo: Cannot locate rosdep definition for [gmapping]
slam_sim_demo: Cannot locate rosdep definition for [hector_mapping]
`
sudo apt-get install 了他说的三个依赖，都说没有找到。镜像源用的清华的，一直都很好用
`kenneth@kenneth-lenovo-g50-75m:~/catkin_ws$ sudo apt-get install yocs_cmd_vel_mux
[sudo] password for kenneth: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package yocs_cmd_vel_mux
`
不想换环境 没有道理用新版本不行 /(ㄒoㄒ)/

jimsonbuilder, 2018-08-10:
我的问题已经解决，应该不是版本的问题，我安装的版本是indigo，按他提供的pdf讲义说明，我把kinectic版本换成indigo，安装完所有依赖项，问题得到了解决。
我确少的依赖是这个：rospy，因为我当时运行这个节点robot_keyboard_teleop.py，缺少了py依赖无法识别。
讲义的PDF地址是：https://legacy.gitbook.com/book/sychaichangkun/ros-tutorial-icourse163/details，希望对你有帮助。
不行的话把工作空间删了，软件包删了重新来操作一遍吧，good luck to you。

Send time: Friday, Aug 10, 2018 11:05 AM
To: "DroidAITech/ROS-Academy-for-Beginners"<ROS-Academy-for-Beginners@noreply.github.com>; 
Cc: "朱俊程"<870255589@qq.com>; "Comment"<comment@noreply.github.com>;

huiminglhm, 2018-09-26:
我目前也是这个错误，也不想换环境，请问有解决办法吗？

1miaoting1, 2018-10-10:
折腾了一下午，重装ROS，换源，一个一个装依赖。依赖的问题解决了。cmake又报错。难受。

Okabezhu, 2019-04-14:
请问如果ros版本是melodic可以运行程序吗？添加依赖相的时候总是报错

zhujiale@zhujiale-ThinkPad-E550:~/catkin_ws$ rosdep    install    --from-paths    src    --ignore-src    --rosdistro=melodic    -y 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
simple_markers: Cannot locate rosdep definition for [visulization_msgs]
navigation_sim_demo: Cannot locate rosdep definition for [gmapping]
slam_sim_demo: Cannot locate rosdep definition for [hector_mapping]

ysgclight, 2019-05-01:
执行代码：
cd catkin_ws/src
git clone https://github.com/ros-perception/slam_gmapping.git
git clone https://github.com/ros-perception/openslam_gmapping.git
git clone https://github.com/tu-darmstadt-ros-pkg/hector_slam.git
cd ..
rosdep install  --from-paths src --ignore-src
catkin_make

然后出错在：
-- +++ processing catkin package: '**slam_sim_demo**'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/slam_sim_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find hector_mapping (missing: hector_mapping_DIR)
-- Could not find the required component 'hector_mapping'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "hector_mapping"
  with any of the following names:

    hector_mappingConfig.cmake
    hector_mapping-config.cmake

  Add the installation prefix of "hector_mapping" to CMAKE_PREFIX_PATH or set
  "hector_mapping_DIR" to a directory containing one of the above files.  If
  "hector_mapping" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/slam_sim_demo/CMakeLists.txt:3 (find_package)

nTnZone, 2019-07-10:
> 执行代码：
> cd catkin_ws/src
> git clone https://github.com/ros-perception/slam_gmapping.git
> git clone https://github.com/ros-perception/openslam_gmapping.git
> git clone [...]
我也是在CMAKE这里卡住了，请问同学你最后解决了吗

MacrenBB, 2019-07-10:
这个我也没有解决------------------ 原始邮件 ------------------

 

 
  
&mdash;

anchuanxu, 2019-07-15:
## 这个需要改两处:

### 1.下载melodic未发布的两个包到`catkin_ws/src`下:

```
git clone https://github.com/ros-perception/slam_gmapping.git
git clone https://github.com/tu-darmstadt-ros-pkg/hector_slam.git
```
### 2.修改ROS-Academy-for-Beginners/slam_sim_demo下的package.xml，

增加：

```
<build_depend>gmapping</build_depend>
<build_depend>hector_mapping</build_depend>

<run_depend>gmapping</run_depend>
<run_depend>hector_mapping</run_depend>
```

### 再次安装依赖:

```
 rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
```
则成功安装:
```
#All required rosdeps installed successfully
```
##### PS:课程的讲义和代码包会在两周内完成新版本(melodic)的升级,欢迎关注.

@jimsonbuilder @KennethYangle @huiminglhm @1miaoting1

yaoif, 2019-07-20:
> ## 这个需要改两处：
> ### 1.下载旋律发布未两个的包到`catkin_ws/src`下：
> ```
> git clone https://github.com/ros-perception/slam_gmapping.git
> git clone https://github.com/tu-darmstadt-ros-pkg/hector_slam.git
> git clone https://github.com/ros-perception/openslam_gmapping.git 
 
> 
> ### 2.修改ROS-学院换初学者/ slam_sim_demo下的package.xml中，
> 增加：
> 
> ```
> <build_depend>gmapping</build_depend>
> <build_depend>hector_mapping</build_depend>
> 
> <run_depend>gmapping</run_depend>
> <run_depend>hector_mapping</run_depend>
> ```
> 
> ### 再次安装依赖：
> ```
>  rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
> ```
> 
> 则成功安装：
> 
> ```
> #All required rosdeps installed successfully
> ```
> 
> ##### PS：课程的讲义和代码包会在两周内完成新版本（旋律）的升级，欢迎关注。
> @jimsonbuilder @KennethYangle @huiminglhm @ 1miaoting1 @MacrenBB @nTnZone

####################现在是catkin_make的时候出错###############################
@@@@@@@@@@@@@@@
修改ROS-学院换初学者/ slam_sim_demo下的package.xml中，
增加：

<build_depend>gmapping</build_depend>
<build_depend>hector_mapping</build_depend>

然后catkin_make就没有问题

Theigrams, 2019-07-27:
> ## 这个需要改两处:
> ### 1.下载melodic未发布的两个包到`catkin_ws/src`下:
> ```
> git clone https://github.com/ros-perception/slam_gmapping.git
> git clone https://github.com/tu-darmstadt-ros-pkg/hector_slam.git
> ```
> 
> ### 2.修改ROS-Academy-for-Beginners/slam_sim_demo下的package.xml，
> 增加：
> 
> ```
> <build_depend>gmapping</build_depend>
> <build_depend>hector_mapping</build_depend>
> 
> <run_depend>gmapping</run_depend>
> <run_depend>hector_mapping</run_depend>
> ```
> 
> ### 再次安装依赖:
> ```
>  rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
> ```
> 
> 则成功安装:
> 
> ```
> #All required rosdeps installed successfully
> ```
> 
> ##### PS:课程的讲义和代码包会在两周内完成新版本(melodic)的升级,欢迎关注.
> @jimsonbuilder @KennethYangle @huiminglhm @1miaoting1 @MacrenBB @nTnZone

亲测有用，非常感谢老师

Rorywh, 2019-07-31:
My melodic Rod still shows the CMAKE error with the hector_mapping after using above measures. 
Here is what's said on the "ros.org" concerning this problem: Apart from that, coming to the main error, hector_mapping package hasn`t been released for melodic yet.

brucetend, 2019-11-05:
老师您好！请问，课程的讲义和代码包会在两周内完成新版本(melodic)的升级在哪里寻找呀！

qiuzhichang, 2019-12-05:
> 同样的问题，真的没有解决办法吗
> rosdep install --from-paths src --ignore-src --rosdistro melodic -y
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> robot_sim_demo: Cannot locate rosdep definition for [yocs_cmd_vel_mux]
> slam_sim_demo: Cannot locate rosdep definition for [hector_mapping]

-------------------------------------------
-------------------------------------------

sudo apt-get install ros-melodic-yocs-cmd-vel-mux
 这样子

suc1, 2021-09-15:
Download [Melodic (ROS1) VB](https://rosi-images.datasys.swri.edu/)
The ubuntu version 18.04.6 (bionic bear), Get errors:
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> hector_mapping: No definition of [boost] for OS version [bionic]
>

laohuijiadezhu, 2021-10-30:
我安的kinetic也是这个错误

18271404400xxx, 2023-08-14:
[WARN] [1691999999.537353, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.根本进不去，一直显示等待加载模型后跳出这个