DroidAITech/ROS-Academy-for-Beginners #15: 编译时出错

RISwang, 2018-08-08:
以下是编译时终端的报错。
```
[ 98%] Building CXX object ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/src/dishes_Client.cpp.o
In file included from /opt/ros/kinetic/include/actionlib/managed_list.h:38:0,
                 from /opt/ros/kinetic/include/actionlib/client/client_helpers.h:47,
                 from /opt/ros/kinetic/include/actionlib/client/action_client.h:38,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:47,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Client.cpp:2:
/opt/ros/kinetic/include/actionlib/destruction_guard.h: In member function ‘void actionlib::DestructionGuard::destruct()’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(float)’
       count_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0f));
                                                                                ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:38,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Client.cpp:2:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<T>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<T>, void>::type*) [with T = float]’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<float>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:38,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Client.cpp:2:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                              ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&’
In file included from /opt/ros/kinetic/include/actionlib/server/handle_tracker_deleter.h:42:0,
                 from /opt/ros/kinetic/include/actionlib/server/action_server.h:51,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:42,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Server.cpp:2:
/opt/ros/kinetic/include/actionlib/destruction_guard.h: In member function ‘void actionlib::DestructionGuard::destruct()’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(float)’
       count_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0f));
                                                                                ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Server.cpp:2:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<T>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<T>, void>::type*) [with T = float]’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80:   required from here
[... 92 lines cut ...]
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/wangtianfu/catkin_ws_wtf/src/ROS-Academy-for-Beginners/action_demo/src/dishes_Server.cpp:2:
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h: In instantiation of ‘void actionlib::SimpleActionServer<ActionSpec>::executeLoop() [with ActionSpec = action_demo::DoDishesAction_<std::allocator<void> >]’:
/opt/ros/kinetic/include/actionlib/server/simple_action_server_imp.h:111:53:   required from ‘actionlib::SimpleActionServer<ActionSpec>::SimpleActionServer(ros::NodeHandle, std::__cxx11::string, actionlib::SimpleActionServer<ActionSpec>::ExecuteCallback, bool) [with ActionSpec = action_demo::DoDishesAction_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>; actionlib::SimpleActionServer<ActionSpec>::ExecuteCallback = boost::function<void(const boost::shared_ptr<const action_demo::DoDishesGoal_<std::allocator<void> > >&)>; typename ActionSpec::_action_go [... cut ...]
ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/build.make:62: recipe for target 'ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/src/dishes_Client.cpp.o' failed
make[2]: *** [ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/src/dishes_Client.cpp.o] Error 1
CMakeFiles/Makefile2:7273: recipe for target 'ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/all' failed
make[1]: *** [ROS-Academy-for-Beginners/action_demo/CMakeFiles/dishes_Client.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

elevenjiang1, 2018-12-14:
我也是这样子的问题,是编译出错了,也不知道为什么,在git clone下载了Beginners的文件之后编译不通过,报错的error就是这个

RISwang, 2018-12-15:
虽然编译出错但是可以运行，我也就没有管。

closed 2023-10-11 by tfwang08