DroidAITech/ROS-Academy-for-Beginners #16: catkin_make时出错

Yijun-Mao, 2018-08-09:
Base path: /home/maoyijun/ws
Source space: /home/maoyijun/ws/src
Build space: /home/maoyijun/ws/build
Devel space: /home/maoyijun/ws/devel
Install space: /home/maoyijun/ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/maoyijun/ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/maoyijun/ws/build"
####
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Linking CXX executable /home/maoyijun/ws/devel/lib/param_demo/param_demo
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _service_demo_generate_messages_check_deps_Greeting
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target _topic_demo_generate_messages_check_deps_gps
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesResult
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesFeedback
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesActionResult
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesActionFeedback
CMakeFiles/param_demo.dir/src/param_demo.cpp.o：在函数‘main’中：
param_demo.cpp:(.text+0x70)：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
param_demo.cpp:(.text+0xcc)：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, [  0%] Built target actionlib_msgs_generate_messages_lisp
std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
param_demo.cpp:(.text+0x12d)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x19b)：对‘ros::NodeHandle::getParam(std::string const&, int&) const’未定义的引用
param_demo.cpp:(.text+0x2ae)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string[  0%] Built target actionlib_msgs_generate_messages_nodejs
 const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x3c8)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x4e2)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x5fc)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x6e0)：对‘ros::NodeHandle::hasParam(std::string const&) const’未定义的引用
param_demo.cpp:(.text+0x774)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x88e)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x97a)：对‘ros::param::set(std::string const&, int)’未定义的引用
param_demo.cpp:(.text+0x9eb)：对‘ros::NodeHandle::setParam(std::string const&, int) const’未定义的引用
param_demo.cpp:(.text+0xa75)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0xb85)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0xc6c)：对‘ros::NodeHandle::hasParam(std::string const&) const’未定义的引用
param_demo.cpp:(.text+0xcc9)：对‘ros::param::has(std::string const&)’未定义的引用
param_demo.cpp:(.text+0xd66)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0xe7d)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0xf97)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x10ae)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1192)：对‘ros::NodeHandle::deleteParam(std::string const&) const’未定义的引用
param_demo.cpp:(.text+0x11ef)：对‘ros::param::del(std::string const&)’未定义的引用
param_demo.cpp:(.text+0x128c)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x13a3)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x14bd)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x15d4)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1718)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x17fc)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x1890)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1979)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x1a0d)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1af6)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x1b8a)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1c73)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x1d07)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1df0)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x1e84)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
param_demo.cpp:(.text+0x1f6d)：对‘ros::param::get(std::string const&, int&)’未定义的引用
param_demo.cpp:(.text+0x2001)：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/param_demo.dir/src/param_demo.cpp.o：在函数‘bool ros::NodeHandle::param<int>(std::string const&, int&, int const&) const’中：
param_demo.cpp:(.text._ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_]+0x27)：对‘ros::NodeHandle::hasParam(std::string const&) const’未定义的引用
param_demo.cpp:(.text._ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_]+0x42)：对‘ros::NodeHandle::getParam(std::string const&, int&) const’未定义的引用
collect2: error: ld returned 1 exit status
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesAction
[  0%] Built target actionlib_msgs_generate_messages_cpp
ROS-Academy-for-Beginners/param_demo/CMakeFiles/param_demo.dir/build.make:113: recipe for target '/home/maoyijun/ws/devel/lib/param_demo/param_demo' failed
make[2]: *** [/home/maoyijun/ws/devel/lib/param_demo/param_demo] Error 1
CMakeFiles/Makefile2:975: recipe for target 'ROS-Academy-for-Beginners/param_demo/CMakeFiles/param_demo.dir/all' failed
make[1]: *** [ROS-Academy-for-Beginners/param_demo/CMakeFiles/param_demo.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesActionGoal
[  0%] Built target _action_demo_generate_messages_check_deps_DoDishesGoal
[  0%] Built target _msgs_demo_generate_messages_check_deps_AutoDockingFeedback
[  0%] Built target _msgs_demo_generate_messages_check_deps_Trigger
[  0%] Built target _msgs_demo_generate_messages_check_deps_Empty
[  0%] Built target _msgs_demo_generate_messages_check_deps_SetBool
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

——————————————————————————————————————————————
我安装的是kinetic版本，按照README进行操作，在catkin_make后出现这些报错，请问这是怎么回事？

sychaichangkun, 2018-08-12:
链接问题，首先确保你ROS安装并且正常运行，然后看下param_demo包里面的CMakeLists.txt，看看是不是少了什么

hangxiaotian, 2018-08-18:
ubuntu18.04版本可以吗，老师？

1miaoting1, 2018-10-10:
老师，是不是18.04不太适用教程啊？

wykxwyc, 2018-12-02:
> @wykxwyc open your CMakeLists.txt file and make sure the executable has been linked to ${CATKIN_LIBRARIES}

Thanks for your advice, and I finally figured out my newly met problem was not same as the problem described above. And the solution to my problem can be found [here](https://stackoverflow.com/questions/37004976/undefined-reference-to-processstd-cxx11basic-string-when-compiling-a).

Thanks again for your kindness!

WANGshouming4937, 2019-04-19:
这个问题解决了吗

hankerbit, 2020-07-01:
同样的报错问题  但是可以用