DroidAITech/ROS-Academy-for-Beginners #19: 运行仿真程序时出错

commencal, 2018-09-10:
bbb@ubuntu:~$ rospack profile
Full tree crawl took 1.718104 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
1.588017   /opt/ros/kinetic/share
0.128410   /home/bbb/tutorial_ws/src
0.128268   /home/bbb/tutorial_ws/src/ROS-Academy-for-Beginners
0.003235 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev
0.000887 * /opt/ros/kinetic/share/doc
0.000158 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev/haarcascades
0.000069 * /opt/ros/kinetic/share/OpenCV-3.3.1-dev/lbpcascades
0.000028 * /opt/ros/kinetic/share/doc/liborocos-kdl
bbb@ubuntu:~$ roslaunch robot_sim_demo robot_spawn.launch
... logging to /home/bbb/.ros/log/a562eb50-b508-11e8-8c14-000c295ac99e/roslaunch-ubuntu-14387.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://ubuntu:44121/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/bbb/tutoria...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [14402]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a562eb50-b508-11e8-8c14-000c295ac99e
process[rosout-1]: started with pid [14415]
started core service [/rosout]
process[gazebo-2]: started with pid [14418]
process[gazebo_gui-3]: started with pid [14423]
[... 41 lines cut ...]
[ INFO] [1536590953.692240324, 0.001000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1536590953.694856759, 0.001000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1536590953.712755582, 0.001000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1536590953.716938612, 0.001000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1536590953.862291655, 0.001000000]: Loading gazebo_ros_control plugin
[ INFO] [1536590953.862505226, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /xbot
[ INFO] [1536590953.863654989, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1536590954.072101571, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'base_to_yaw_platform'.
[ WARN] [1536590954.074140768, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'yaw_to_pitch_platform'.
[ INFO] [1536590954.091793080, 0.001000000]: Loaded gazebo_ros_control.
[ INFO] [1536590954.151134094, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1536590954.153255156, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1536590954.235731, 0.047000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1536590954.241772, 0.049000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1536590954.246240, 0.049000]: Loading controller: joint_state_controller
[INFO] [1536590954.330414, 0.066000]: Loading controller: yaw_platform_position_controller
[ INFO] [1536590954.479945682, 0.108000000]: Physics dynamic reconfigure ready.
[ INFO] [1536590954.496309231, 0.111000000]: Physics dynamic reconfigure ready.
[INFO] [1536590954.576465, 0.132000]: Loading controller: pitch_platform_position_controller
[INFO] [1536590954.688976, 0.161000]: Controller Spawner: Loaded controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller
[INFO] [1536590954.694475, 0.162000]: Started controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller
VMware: vmw_ioctl_command error Invalid argument.
Aborted (core dumped)
[gazebo_gui-3] process has died [pid 14423, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/bbb/.ros/log/a562eb50-b508-11e8-8c14-000c295ac99e/gazebo_gui-3.log].
log file: /home/bbb/.ros/log/a562eb50-b508-11e8-8c14-000c295ac99e/gazebo_gui-3*.log

jakalaka, 2018-09-17:
Did u solve the issue? I got the same XML Element[mu] not defined in SDF. Ignoring[mu] while using melodic

commencal, 2018-09-17:
I havent sloved this problem.I don't know if it's because I'm using virtual machines to cause errors

YCH188, 2019-04-11:
I have the same problem, and it is not caused by virtual machines. Have you solved this?

YCH188, 2019-04-11:
@jakalaka @commencal