DroidAITech/ROS-Academy-for-Beginners #37: 运行robot_sim_demo只加载出了地板和一个台子

tinytinywell, 2019-05-05:
[image]

[image]

实际上这个台子也没有加载完全
不知道是不是因为是在虚拟机里运行的缘故
请问老师这个问题要怎么解决呢?
在不装双系统的前提下能否解决呢？

YCH188, 2019-05-12:
@tinytinywell 你的gazebo版本应该比较高，换成7.0就解决了，建议装个双系统，做slam还是比较耗内存的

tinytinywell, 2019-05-15:
> 
> 
> @tinytinywell 你的gazebo版本应该比较高，换成7.0就解决了，建议装个双系统，做slam还是比较耗内存的

谢谢前辈回复
请问一下前辈对cartographer有了解吗

YCH188, 2019-05-18:
@tinytinywell 我本身不是搞SLAM的，对这个不了解

xiaozaiziwyt, 2019-05-18:
你在另一个端口继续输入 `rosrun robot_sim_demo robot_keyboard_teleop.py` 
提示 robot_sim_demo not  found吗？

xiaozaiziwyt, 2019-05-18:
> @tinytinywell 你的gazebo版本应该比较高，换成7.0就解决了，建议装个双系统，做slam还是比较耗内存的

怎么降低这个的版本呢？ 直接卸载重装吗？

tinytinywell, 2019-05-18:
> 
> 
> 你在另一个端口继续输入 `rosrun robot_sim_demo robot_keyboard_teleop.py`
> 提示 robot_sim_demo not found吗？

没有试过哦，因为我后来直接重装了ubuntu系统安装了7.x版本的gazebo然后就有模型了

Theigrams, 2019-07-27:
看一下这个链接
https://q.droid.ac.cn/t/topic/1048

anchuanxu, 2019-08-08:
代码已更新，请切换代码仓库分支到melodic重新下载编译即可。

XZZZ9, 2022-03-09:
您好问题解决了吗 我也是这个问题gazebo11