DroidAITech/ROS-Academy-for-Beginners #39: The problem of rosdep

geskliu, 2019-05-15:
administator@administator-Lenovo:~/catkin_ws$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:

roch_concert: Cannot locate rosdep definition for [concert_service_indoor_2d_map_prep]

When I run the 'rosdep' order , It will show the above error . Please help me to solve the problem , Thanks !

geskliu, 2019-05-15:
I had installed the roch robot to study the ros . The error message is about the roch . Must I uninstall the roch robot to install the dependcies for x-bot ? But I don't wanner uninstall the roch . What should I do ?

Ronghua19911, 2019-06-25:
先试一下sudo apt-get update,如果有更新报错，就有可能是你ros的key有问题，
你可以尝试http://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/
上的解决方法