DroidAITech/ROS-Academy-for-Beginners #44: catkin_make错误

zoroppace, 2019-07-08:
请老师指教
[image]

Baymax94, 2019-07-08:
同学，可以参考一下：  
我遇到过，也是catkin_make失败，是因为这一步  
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y  
有的包没有安装上，导致后面失败  
查到说可以忽略这个包，重新catkin_make  
我用的是重新运行所有依赖包安装命令，有修改：  
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic  
安装过程，自己手动输入y

zoroppace, 2019-07-08:
谢谢！！

Sung-Hoiank, 2019-08-07:
> 同学，可以参考一下：
> 我遇到过，也是catkin_make失败，是因为这一步
> $ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
> 有的包没有安装上，导致后面失败
> 查到说可以忽略这个包，重新catkin_make
> 我用的是重新运行所有依赖包安装命令，有修改：
> $ rosdep install --from-paths src --ignore-src --rosdistro=kinetic
> 安装过程，自己手动输入y
实测有效，十分感谢！搞了挺久终于搞出来了

Sung-Hoiank, 2019-08-07:
感谢Baymax94！ 实测有效！

sunziyang-hi, 2021-01-07:
解决了，真的太给力了，感谢