DroidAITech/ROS-Academy-for-Beginners #45: 运行robot_sim_demo没有地图，只有地板

zouzhe1, 2019-07-10:
roslaunch robot_sim_demo robot_spawn.launch

[image]
[image]

zouzhe1, 2019-07-10:
这是要等自动下载地图？我等好几个小时也没有反映，需要vpn吗？

anchuanxu, 2019-08-08:
已经更新了代码，切换到melodic分支，重新下载编译即可

zouzhe1, 2019-08-20:
@anchuanxu   你说的melodic分支啥意思？是这个吗git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git
这个里面只有讲义，没有代码。
还是说，你说的分支是要我把这句改成melodic？
rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y

anchuanxu, 2019-08-20:
@zouzhe1 不是不是，https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic　直接clone这个链接就是兼容Ubuntu18的代码仓库了。

anchuanxu, 2019-08-20:
@zouzhe1 melodic分支是课程代码仓库的一个分支，仓库有多个分支 分别支持不同版本的ROS，下载对应版本代码就可以避免版本不兼容的问题。

zouzhe1, 2019-08-20:
@anchuanxu 明白了，谢谢.

zouzhe1, 2019-08-20:
@anchuanxu 非常感谢，已经成功运行，第一次玩ros，看到这个还是很兴奋的。还想请问一下，你们这个场景是用什么作的？我想把我家也做成这种三维的gazebo，应该用什么软件呢？设备上用什么相机呢？

zouzhe1, 2019-10-20:
@Joker621  我弄好了，不是分支，是另一个专门的ubuntu18的。

ZehuaJia, 2019-11-15:
> @Joker621 我弄好了，不是分支，是另一个专门的ubuntu18的。

请问怎么搞定的，我下载了最新链接里melodic分支的内容，仍然在gazebo里和你的图一样加载不出来模型

c66zwzly, 2020-02-18:
我换成melodic版本的package的之后更新了一下gazebo，结果好像出了点问题，又update upgrade了一下，然后重新装了gazebo和依赖，打开就正常了

518021910636, 2020-04-15:
> 我换成melodic版本的package的之后更新了一下gazebo，结果好像出了点问题，又update upgrade了一下，然后重新装了gazebo和依赖，打开就正常了

请问如何重装gazebo和依赖，gazebo9，换melodic的代码包后碰到同样的问题，只有地板和台子

anchuanxu, 2020-04-15:
> > 我换成melodic版本的package的之后更新了一下gazebo，结果好像出了点问题，又update upgrade了一下，然后重新装了gazebo和依赖，打开就正常了
> 
> 请问如何重装gazebo和依赖，gazebo9，换melodic的代码包后碰到同样的问题，只有地板和台子

你好，首先确认系统版本是ubuntu18下载安装玩ros-melodic后,请下载https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic 该链接下代码进行编译。换gazebo等均在readme中有提及。请尝试下看看。

518021910636, 2020-04-15:
> > > 我换成melodic版本的package的之后更新了一下gazebo，结果好像出了点问题，又update upgrade了一下，然后重新装了gazebo和依赖，打开就正常了
> > 
> > 
> > 请问如何重装gazebo和依赖，gazebo9，换melodic的代码包后碰到同样的问题，只有地板和台子
> 
> 你好，首先确认系统版本是ubuntu18下载安装完ros-melodic后,请下载https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic 该链接下代码进行编译。换gazebo等均在readme中有提及。请尝试下看看。
已解决，感谢老师！
ubuntu18.04,gazebo9,melodic,新的代码是完全没有问题的，但是git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git后默认是在master分支上，要手动调整分支到melodic下。具体操作方法我看老师提到但是没有具体指引，下面附上解决方式供参考。
$ git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git
$ cd ~/path/catkin_ws/src/ROS-Academy-for-Beginners
$ git branch -r
$ git checkout origin/melodic
$ git checkout -b origin/melodic
$ cd ~/path/catkin_ws
$ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
$ source ~/2020.4.4/ros3/catkin_ws/devel/setup.bash
$ roslaunch robot_sim_demo robot_spawn.launch
这样成功解决！

anchuanxu, 2020-04-16:
> > > > 我换成melodic版本的package的之后更新了一下gazebo，结果好像出了点问题，又update upgrade了一下，然后重新装了gazebo和依赖，打开就正常了
> > > 
> > > 
> > > 请问如何重装gazebo和依赖，gazebo9，换melodic的代码包后碰到同样的问题，只有地板和台子
> > 
> > 
> > 你好，首先确认系统版本是ubuntu18下载安装完ros-melodic后,请下载https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic 该链接下代码进行编译。换gazebo等均在readme中有提及。请尝试下看看。
> 
> 已解决，感谢老师！
> ubuntu18.04, gazebo9, melodic
> 新的代码是完全没有问题的，但是
> git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git
> 之后默认是在master分支上，要手动调整分支到melodic下。具体操作方法我看老师提到但是没有具体指引，下面附上解决方式供参考。
> $ git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git
> $ cd ~/path/catkin_ws/src/ROS-Academy-for-Beginners
> $ git branch -r
> $ git checkout origin/melodic
> $ git checkout -b origin/melodic
> $ cd ~/path/catkin_ws
> $ rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
> $ source ~/2020.4.4/ros3/catkin_ws/devel/setup.bash
> $ roslaunch robot_sim_demo robot_spawn.launch
> 这样成功解决！

good! 切换到melodic分支上后，编译新的代码基本都没问题了，欢迎提出新的bug和pr!

SSmarshal, 2020-07-12:
请问是在gazebo9上运行不了吗，必须要将版本换成7.x才能使用吗？我换到melodic分支上仍然打开只有地板

SSmarshal, 2020-07-12:
> 请问是在gazebo9上运行不了吗，必须要将版本换成7.x才能使用吗？我换到melodic分支上仍然打开只有地板

不好意思，我刚刚没在意，我以为可以只把分支clone到码云上的。。。还是要选择一下

linghushaoxia-wqw, 2020-09-26:
请问需要把原来的包删掉吗？系统提示 fatal: destination path 'ROS-Academy-for-Beginners' already exists and is not an empty directory.

linghushaoxia-wqw, 2020-09-26:
已解决   安装完后会自动删除旧的包

linchen2020, 2021-01-30:
> good! 切换到melodic分支上后，编译新的代码基本都没问题了，欢迎提出新的bug和pr!

真的很牛，我用的是TX2下的ubuntu18.04  ros melodic ，刚开始出现地板，按照这样做就解决了问题，非常感谢你们提供帮助

nightsky-nostar, 2021-02-28:
我的ubuntu18.04 ros melodic是将 https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic换为了https://gitee.com/victordoom/ROS-Academy-for-Beginners-melodic，因为第一个地址我无法访问，然后解决了只有地板的问题。

linxilinYA, 2022-05-01:
我想问问后续呢，之前clone了一次，然后用了楼上的方法把链接改为gitee的，然后运行demo的时候报错有名字相同的文件，想知道如何删除以前的