DroidAITech/ROS-Academy-for-Beginners #61: ubuntu16.04 运行robot_sim_demo 出现[urdf_spawner-4] process has died

sakura745, 2019-09-23:
前段时间运行的时候还没有这个状况 后来重做了系统之后 再次运行就出现这类错误了
[urdf_spawner-4] process has died [pid 2195, exit code 1, cmd /opt/ros/kinetic/lib/gazebo_ros/spawn_model -urdf -x 5.0 -y 0.0 -z 0.0 -Y -2.0 -model xbot-u -param robot_description __name:=urdf_spawner __log:=/home/sakura/.ros/log/41f001de-ddd9-11e9-9a1f-38002588a7bb/urdf_spawner-4.log].
log file: /home/sakura/.ros/log/41f001de-ddd9-11e9-9a1f-38002588a7bb/urdf_spawner-4*.log
而且我找不到机器人了

这个错误说的是什么意思啊 怎么解决呢 
我百度了几个问题 有的说是更新gazebo 但我的gazebo就是更新了之后才这样啊
我看README说推荐gazebo 7.9 版本 我找不到7.9啊

sakura745, 2019-09-23:
哦对了 最开始的时候我看
/opt/ros/kinetic/lib/gazebo_ros/spawn_model -urdf -x 5.0 -y 0.0 -z 0.0 -Y -2.0 -model xbot-u -param
这行的-y 是.0(就是这样：-y .0）我以为是格式不对 结果我改成0.0 也就是现在这样  但是还是出错 看来不是格式的问题

anchuanxu, 2019-11-22:
emmm 代码下载的是16.04的版本吗确定?

sakura745, 2019-11-29:
后来我解决这个问题了 我记得是权限原因

Happytreety, 2022-01-13:
Ubuntu 18.04版本也遇到了这个问题，请问怎么修改权限呢？