DroidAITech/ROS-Academy-for-Beginners #63: 运行roslaunch robot_sim_demo robot_spawn.launch后，gazebo里面什么都没有，终端有如下信息

venson0, 2019-09-29:
... logging to /home/robot/.ros/log/e77e3c26-e287-11e9-b956-00e04ccd1e9e/roslaunch-robot-6427.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://robot:38299/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/robot/tutor...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [6440]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e77e3c26-e287-11e9-b956-00e04ccd1e9e
process[rosout-1]: started with pid [6453]
started core service [/rosout]
process[gazebo-2]: started with pid [6472]
process[gazebo_gui-3]: started with pid [6482]
process[urdf_spawner-4]: started with pid [6487]
process[xbot/spawner-5]: started with pid [6488]
process[xbot/robot_state_publisher-6]: started with pid [6489]
process[mobile_base_nodelet_manager-7]: started with pid [6490]
process[cmd_vel_mux-8]: started with pid [6491]
[INFO] [1569740895.869034, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1569740895.900993908]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1569740895.901617796]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1569740895.941129984]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1569740895.941725516]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
SpawnModel script started
[INFO] [1569740896.214062, 0.000000]: Loading model XML from ros parameter
[INFO] [1569740896.217564, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[WARN] [1569740926.001138, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[xbot/spawner-5] process has finished cleanly
log file: /home/robot/.ros/log/e77e3c26-e287-11e9-b956-00e04ccd1e9e/xbot-spawner-5*.log

venson0, 2019-10-08:
跟显卡品牌有关吗？我的是amd的

tiny138, 2019-10-29:
我用：
killall gzserver
 
kilall gzclient
命令能加载gazebo模型了

anchuanxu, 2019-11-22:
OK good ,

luxingyang, 2020-03-19:
我ubuntu18.04和melodic就遇到这样的问题，仔细发现已经有适合melodic的版本了，可以在ROS_Academy_for_Beginners目录下
切换分支：
git checkout melodic
git pull
再运行就正常了