DroidAITech/ROS-Academy-for-Beginners #70: melodic 版本  roslaunch 之后 gazebo只有地板没有模型

kids0cn, 2020-01-09:
分支正确，编译没有报错，系统环境 18.04  melodic 

运行之后输出如下：

... logging to /home/kids/.ros/log/42e18ae6-32d8-11ea-b80c-34f39af483a5/roslaunch-kids-ThinkPad-X1-Yoga-1st-5860.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://kids-ThinkPad-X1-Yoga-1st:35219/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/kids/tmp/ca...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [5873]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 42e18ae6-32d8-11ea-b80c-34f39af483a5
process[rosout-1]: started with pid [5884]
started core service [/rosout]
process[gazebo-2]: started with pid [5891]
process[gazebo_gui-3]: started with pid [5896]
process[urdf_spawner-4]: started with pid [5901]
process[xbot/spawner-5]: started with pid [5902]
process[xbot/robot_state_publisher-6]: started with pid [5903]
process[mobile_base_nodelet_manager-7]: started with pid [5904]
process[cmd_vel_mux-8]: started with pid [5905]
[INFO] [1578571502.082011, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1578571502.158143125]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1578571502.159660881]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1578571502.284059219]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1578571502.285494542]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Err] [REST.cc:205] Error in REST request

libcurl: (7) Failed to receive SOCKS4 connect request ack.
[INFO] [1578571502.777673, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1578571502.794862, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1578571502.799300, 0.000000]: Calling service /gazebo/spawn_urdf_model
Warning [parser.cc:950] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[ INFO] [1578571503.807325023, 0.001000000]: <robotNamespace> set to: //
[ INFO] [1578571503.807419105, 0.001000000]: <topicName> set to: //imu
[ INFO] [1578571503.807440740, 0.001000000]: <frameName> set to: imu_link
[ INFO] [1578571503.807474102, 0.001000000]: <updateRateHZ> set to: 10
[ INFO] [1578571503.807494135, 0.001000000]: <gaussianNoise> set to: 0
[ INFO] [1578571503.807520303, 0.001000000]: <xyzOffset> set to: 0 0 0
[ INFO] [1578571503.807558355, 0.001000000]: <rpyOffset> set to: 0 -0 0
[ INFO] [1578571504.040034272, 0.001000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1578571504.040107910, 0.001000000]: Starting GazeboRosLaser Plugin (ns = /)
[ INFO] [1578571504.041643590, 0.001000000]: GPU Laser Plugin (ns = /) <tf_prefix_>, set to ""
[ INFO] [1578571504.042476430, 0.001000000]: LoadThread function completed
[INFO] [1578571504.055999, 0.001000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1578571504.090989779, 0.001000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1578571504.095305489, 0.001000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1578571504.156967671, 0.001000000]: Starting plugin DiffDrive(ns = //)
[ WARN] [1578571504.157234527, 0.001000000]: DiffDrive(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1578571504.158039409, 0.001000000]: DiffDrive(ns = //): <tf_prefix> = 
[ WARN] [1578571504.158282280, 0.001000000]: DiffDrive(ns = //): missing <publishOdomTF> default is true
[ WARN] [1578571504.158471439, 0.001000000]: DiffDrive(ns = //): missing <wheelTorque> default is 5
[ WARN] [1578571504.158664612, 0.001000000]: DiffDrive(ns = //): missing <odometrySource> default is 1
[ INFO] [1578571504.159503202, 0.001000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1578571504.160257719, 0.001000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1578571504.162634203, 0.001000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1578571504.163354633, 0.001000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1578571504.188450248, 0.001000000]: Loading gazebo_ros_control plugin
[ INFO] [1578571504.188655467, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /xbot
[ INFO] [1578571504.190291506, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1578571504.308767168, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'base_to_yaw_platform'.
[ WARN] [1578571504.309700265, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'yaw_to_pitch_platform'.
[ INFO] [1578571504.314621624, 0.001000000]: Loaded gazebo_ros_control.
[ INFO] [1578571504.339438301, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[urdf_spawner-4] process has finished cleanly
log file: /home/kids/.ros/log/42e18ae6-32d8-11ea-b80c-34f39af483a5/urdf_spawner-4*.log
[ INFO] [1578571504.390425720, 0.068000000]: Physics dynamic reconfigure ready.
[INFO] [1578571504.495318, 0.170000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1578571504.498968, 0.172000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1578571504.502310, 0.176000]: Loading controller: joint_state_controller
[INFO] [1578571504.523185, 0.190000]: Loading controller: yaw_platform_position_controller
[INFO] [1578571504.595226, 0.248000]: Loading controller: pitch_platform_position_controller
[INFO] [1578571504.639813, 0.274000]: Controller Spawner: Loaded controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller
[INFO] [1578571504.646133, 0.278000]: Started controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller

kids0cn, 2020-01-10:
18.04+melodic,git clone下来master之后，需要通过git checkout melodic  选择melodic分支，再编译，问题解决。

closed 2020-01-10 by kids0cn

leonchan123, 2020-03-16:
Thanks for your mention.
After downloading the package by`git clone https://github.com/DroidAITech/ROS-Academy-for-Beginners.git`,you go into it and run a terminal with`git check melodic`.Then you can catkin_make.
But I met another problem about Qt version,which need Qt 4.x. This is my solution  as follow:
`rosdep install --from-paths WORKSPACE --ignore-src --rosdistro=ROSDISTRO`
By this way,you can deal with the dependency problem. (The WORKSPACE is your catkin workspace,and the ROSDISTRO is kinetic or melodic)