DroidAITech/ROS-Academy-for-Beginners #72: melodic版本运行功能正常，gazebo模型显示不完整

zxyxm, 2020-02-16:
melodic版本运行功能正常，gazebo模型显示不完整

anchuanxu, 2020-02-16:
有截图看看吗？ 确定下载代码是melodic版本的话，理论上模型没问题

zxyxm, 2020-02-17:
[image: 2020-02-17 09-49-41 的屏幕截图.png]
显示不完整，运行：
￥ roslaunch robot_sim_demo robot_spawn.launch

... logging to
/home/yxh/.ros/log/30397876-5124-11ea-9326-f894c26b08eb/roslaunch-yxh-GL62M-7RDX-15339.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://yxh-GL62M-7RDX:40635/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/yxh/catkin_...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [15358]
process[gazebo_gui-2]: started with pid [15363]
process[urdf_spawner-3]: started with pid [15368]
process[xbot/spawner-4]: started with pid [15369]
process[xbot/robot_state_publisher-5]: started with pid [15370]
process[mobile_base_nodelet_manager-6]: started with pid [15371]
process[cmd_vel_mux-7]: started with pid [15374]
[INFO] [1581904081.774679, 0.000000]: Controller Spawner: Waiting for
service controller_manager/load_controller
[INFO] [1581904087.059848, 0.000000]: Loading model XML from ros parameter
robot_description
[INFO] [1581904087.086588, 0.000000]: Waiting for service
/gazebo/spawn_urdf_model
[ INFO] [1581904090.698960463]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1581904090.702481373]: waitForService: Service
[/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1581904090.711759904]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1581904090.715956567]: waitForService: Service
[/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1581904093.723378, 0.000000]: Calling service
[... 75 lines cut ...]
[ INFO] [1581904096.703692144, 0.042000000]: Physics dynamic reconfigure
ready.
[INFO] [1581904096.850866, 0.181000]: Controller Spawner: Waiting for
service controller_manager/switch_controller
[INFO] [1581904096.853164, 0.184000]: Controller Spawner: Waiting for
service controller_manager/unload_controller
[INFO] [1581904096.855289, 0.186000]: Loading controller:
joint_state_controller
[INFO] [1581904096.927634, 0.252000]: Loading controller:
yaw_platform_position_controller
[INFO] [1581904096.985174, 0.286000]: Loading controller:
pitch_platform_position_controller
[INFO] [1581904097.021045, 0.323000]: Controller Spawner: Loaded
controllers: joint_state_controller, yaw_platform_position_controller,
pitch_platform_position_controller
[INFO] [1581904097.027518, 0.328000]: Started controllers:
joint_state_controller, yaw_platform_position_controller,
pitch_platform_position_controller
Topic
[/data://world/default/model/xbot-u/plugin/differential_drive_controller/joint_cmd]
is not valid.
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target
host name 'api.ignitionfuel.org'

anchuanxu, 2020-02-17:
看log应该是没有升级gazebo到9.10版本以上。ps，截图没有发出来。

anchuanxu, 2020-02-17:
升级gazebo到gazebo9步骤：

```
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo9
```

zxyxm, 2020-02-17:
😆噢噢噢噢，谢谢老师，我试一下

Chenxing-NEU, 2020-02-17:
老师，我的gazebo升级到9了，版本是melodic，但是在运行robot_sim_demo时仍然加载不了模型，只有一个地板和一个加载不完全的Xbot-u
[image]

Chenxing-NEU, 2020-02-17:
[image]

zxyxm, 2020-02-17:
我也是

xdc1214, 2020-02-18:
> 老师，我的gazebo升级到9了，版本是melodic，但是在运行robot_sim_demo时仍然加载不了模型，只有一个地板和一个加载不完全的Xbot-u
> [image]

没记错的话, 教程里建议下载kinetic版本. 会不会是版本不兼容的问题?

Chenxing-NEU, 2020-02-18:
> […](#)

我出错的原因找到了，还是版本的问题，按照教程指示直接clone过来的还是kinetic版本的程序，你可以试一下直接在https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic 界面下下载zip压缩包，再解压到src文件夹下，安装依赖之后就可以运行了。
[image]

anchuanxu, 2020-02-19:
good! 在ubuntu18.04下正常下载对应的melodic分支代码是不会出现模型问题的哦，这个问题已经fixed。

anchuanxu, 2020-02-19:
@lazyparser 这个已完成，可以关闭。

closed 2020-02-20 by lazyparser