DroidAITech/ROS-Academy-for-Beginners #73: rviz加载camera闪退

xdc1214, 2020-02-18:
gezabo打开robot_sim_demo robot_spawn.launch一切正常, 打开rosrun robot_sim_demo robot_keyboard_teleop.py也能控制小车运动, 终端输入rviz插入robotModel正常, 插入camera闪退. 下面是终端的显示:

[ INFO] [1582020157.570556099]: rviz version 1.11.10
[ INFO] [1582020157.570621371]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1582020157.818123341, 498.502000000]: Stereo is NOT SUPPORTED
[ INFO] [1582020157.818208092, 498.502000000]: OpenGl version: 3 (GLSL 1.3).
[rospack] Warning: no such package media_export
rviz: /usr/include/boost/smart_ptr/shared_ptr.hpp:648：typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = ros::CallbackQueue; typename boost::detail::sp_member_access<T>::type = ros::CallbackQueue*]: 假设 ‘px != 0’ 失败。
已放弃 (核心已转储)

请问哪位有遇到过这种情况吗?

xdc1214, 2020-02-19:
已找到原因: 安装rosdep时出错, 有插件没安装. 原因是之前安装了gazebo, 引起rosdep安装错误. 解决方法: 先gazebo -v找出安装版本, 我是7, 然后sudo apt-get remove gazebo7, 然后sudo apt-get update && sudo apt-get upgrade, 即可正确安装rosdep.

anchuanxu, 2020-02-19:
good~

anchuanxu, 2020-02-19:
@lazyparser 可以关啦这个

closed 2020-02-20 by lazyparser