DroidAITech/ROS-Academy-for-Beginners #75: 给解答一下吧，move_baseConfig.cmake     move_base-config.cmake

xiaobai951, 2020-02-19:
liu@liu-Lenovo-XiaoXin-700-15ISK:~/catkin_ws$ catkin_make
Base path: /home/liu/catkin_ws
Source space: /home/liu/catkin_ws/src
Build space: /home/liu/catkin_ws/build
Devel space: /home/liu/catkin_ws/devel
Install space: /home/liu/catkin_ws/install
####
#### Running command: "cmake /home/liu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/liu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/liu/catkin_ws/install -G Unix Makefiles" in "/home/liu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/liu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/liu/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/liu/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/liu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 15 packages in topological order:
-- ~~  - ros_academy_for_beginners (metapackage)
-- ~~  - name_demo
-- ~~  - orbslam2_demo
-- ~~  - param_demo
-- ~~  - rtabmap_demo
-- ~~  - service_demo
-- ~~  - topic_demo
-- ~~  - action_demo
-- ~~  - msgs_demo
-- ~~  - navigation_sim_demo
-- ~~  - robot_sim_demo
-- ~~  - slam_sim_demo
-- ~~  - tf_demo
-- ~~  - tf_follower
-- ~~  - urdf_demo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'ros_academy_for_beginners'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/ros_academy_for_beginners)
-- +++ processing catkin package: 'name_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/name_demo)
-- +++ processing catkin package: 'orbslam2_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/orbslam2_demo)
-- +++ processing catkin package: 'param_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/param_demo)
-- +++ processing catkin package: 'rtabmap_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/rtabmap_demo)
-- +++ processing catkin package: 'service_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/service_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- service_demo: 0 messages, 1 services
-- +++ processing catkin package: 'topic_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/topic_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- topic_demo: 1 messages, 0 services
-- +++ processing catkin package: 'action_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/action_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- Generating .msg files for action action_demo/DoDishes /home/liu/catkin_ws/src/ROS-Academy-for-Beginners/action_demo/action/DoDishes.action
-- action_demo: 7 messages, 0 services
-- +++ processing catkin package: 'msgs_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/msgs_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action msgs_demo/AutoDocking /home/liu/catkin_ws/src/ROS-Academy-for-Beginners/msgs_demo/action/AutoDocking.action
-- Generating .msg files for action msgs_demo/AddTwoInts /home/liu/catkin_ws/src/ROS-Academy-for-Beginners/msgs_demo/action/AddTwoInts.action
-- Generating .msg files for action msgs_demo/GetMap /home/liu/catkin_ws/src/ROS-Academy-for-Beginners/msgs_demo/action/GetMap.action
-- Generating .msg files for action msgs_demo/MoveBase /home/liu/catkin_ws/src/ROS-Academy-for-Beginners/msgs_demo/action/MoveBase.action
-- msgs_demo: 42 messages, 9 services
-- +++ processing catkin package: 'navigation_sim_demo'
-- ==> add_subdirectory(ROS-Academy-for-Beginners/navigation_sim_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "move_base" with
  any of the following names:

    move_baseConfig.cmake
    move_base-config.cmake

  Add the installation prefix of "move_base" to CMAKE_PREFIX_PATH or set
  "move_base_DIR" to a directory containing one of the above files.  If
  "move_base" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/navigation_sim_demo/CMakeLists.txt:3 (find_package)

-- Could not find the required component 'move_base'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base" with
  any of the following names:

    move_baseConfig.cmake
    move_base-config.cmake

  Add the installation prefix of "move_base" to CMAKE_PREFIX_PATH or set
  "move_base_DIR" to a directory containing one of the above files.  If
  "move_base" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/navigation_sim_demo/CMakeLists.txt:3 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/liu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/liu/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

anchuanxu, 2020-02-19:
包没有装全，直观看是缺少move_base包。可以通过`sudo apt-get install ros-kinetic-move-base`来试试

xiaobai951, 2020-02-20:
> 包没有装全，直观看是缺少move_base包。可以通过`sudo apt-get install ros-kinetic-move-bas解决了，谢谢

xiaobai951, 2020-02-20:
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "map_server" with
  any of the following names:

    map_serverConfig.cmake
    map_server-config.cmake

  Add the installation prefix of "map_server" to CMAKE_PREFIX_PATH or set
  "map_server_DIR" to a directory containing one of the above files.  If
  "map_server" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/navigation_sim_demo/CMakeLists.txt:3 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/liu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/liu/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed