DroidAITech/ROS-Academy-for-Beginners #77: rviz点云坐标出现问题，显示在天上

Hamsters-and-groundhogs, 2020-02-29:
我下载的是melodic分支的ROS-Academy-for-Beginners，跟着视频学习，打开日rvzi的pointcloud2之后，点云的显示在了天上。请问是有代码有bug吗，我应该怎么操作？
[image]

Hamsters-and-groundhogs, 2020-03-02:
你好，我今天在实验Rtabmap时，点云问题直接导致整个地图的搭建十分混乱，而且回环检测也显示没有开启。请问，我应该怎么去调试代码才能解决点云位置错误的问题呢？这个问题困扰了我好久。
[image]

Hamsters-and-groundhogs, 2020-03-02:
你好，我今天在实验Rtabmap时，点云问题直接导致整个地图的搭建十分混乱，而且回环检测也显示没有开启。请问，我应该怎么去调试代码才能解决点云位置错误的问题呢？这个问题困扰了我好久。[image]

Javid-Xi, 2020-03-08:
也有同样问题,求助

wycnet, 2020-05-10:
你解决这个问题了吗，我的rviz点云也是显示在天上。