DroidAITech/ROS-Academy-for-Beginners #80: 老师你好，请教一下关于robot_sim_demo的配置文件出错问题。

KuoYa233, 2020-03-20:
我的系统是18.04lts，ros版本是melodic。~/catkin_ws中成功编译在[链接](https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic)下载的melodic教学资源包。

但是source之后，运行慕课中的演示指令：roslaunch robot_sim_demo robot_spawn.launch
终端打印了以下错误信息。请问怎么解决这个问题？

hairuo@hairuo-Lenovo-Legion-Y7000:~/catkin_ws$ roslaunch robot_sim_demo robot_spawn.launch
... logging to /home/hairuo/.ros/log/277e3ea4-6a99-11ea-90f9-8c16458681e1/roslaunch-hairuo-Lenovo-Legion-Y7000-13525.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

substitution args not supported:  No module named 'rospkg'
when processing file: /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/urdf/robot.xacro
RLException: while processing /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/launch/include/xbot-u.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description_obj]: command [/opt/ros/melodic/share/xacro/xacro.py /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/urdf/robot.xacro] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" name="robot_description_obj"/>
The traceback for the exception was written to the log file

不支持替代参数：没有名为“ rospkg”的模块
处理文件时：/home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/urdf/robot.xacro
RLException：处理/home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/launch/include/xbot-u.launch.xml时：
无效的<param>标记：无法加载命令参数[robot_description_obj]：命令[/opt/ros/melodic/share/xacro/xacro.py / home / hairuo / catkin_ws / src / ROS-Academy-for-Beginners-melodic / robot_sim_demo /urdf/robot.xacro]返回的代码为[2]。

参数xml是<param command =“ $（查找xacro）/xacro.py $（查找robot_sim_demo）/urdf/robot.xacro” name =“ robot_description_obj” />
对该异常的回溯已写入日志文件

以下是robot.xacro的内容：
<?xml version="1.0"?>
<robot xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller"
  xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface"
  xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor"
  xmlns:xacro="http://ros.org/wiki/xacro"
  name="sick_tim">
  
  <xacro:include filename="$(find robot_sim_demo)/urdf/sick_tim.urdf.xacro" />

  
  
  <xacro:rplidarA2 name="laser" ros_topic="scan"   />

  <xacro:include filename="$(find robot_sim_demo)/urdf/materials.xacro" />
  <xacro:include filename="$(find robot_sim_demo)/urdf/xbot-u.urdf.xacro" />
  <xacro:include filename="$(find robot_sim_demo)/urdf/xbot-u.gazebo" />
  <xacro:xbot2_base/>
</robot>

以下是xbot-u.launch.xml的内容：
<launch>

  <arg name="x" default="5.0" />
  <arg name="y" default="1.0" />
  <arg name="z" default="0.0" />
  <arg name="yaw" default="3.1415926" />

  
  

  <param name="robot_description_obj" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" />
  
  <param name="robot_description" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot_dae.xacro" />

  
  <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
        args="-urdf -x $(arg x) -y $(arg y) -z $(arg z) -Y $(arg yaw) -model xbot-u -param robot_description_obj"/>

  

  
  <rosparam file="$(find robot_sim_demo)/param/xbot-u_control.yaml" command="load"/>

  
  <node name="spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" ns="/xbot" args="joint_state_controller
    yaw_platform_position_controller
    pitch_platform_position_controller
    "/>  

  
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" ns="/xbot" respawn="false" output="screen">
    <param name="publish_frequency" value="20.0"/>
  </node>

  
  

  
  

  
  <node pkg="nodelet" type="nodelet" name="mobile_base_nodelet_manager" args="manager"/>
  <node pkg="nodelet" type="nodelet" name="cmd_vel_mux"  args="load yocs_cmd_vel_mux/CmdVelMuxNodelet mobile_base_nodelet_manager">
    <param name="yaml_cfg_file" value="$(find robot_sim_demo)/param/mux.yaml"/>
    <remap from="cmd_vel_mux/output/cmd_vel" to="cmd_vel"/>
  </node>

</launch>

KuoYa233, 2020-03-21:
以下是xbot-u.launch.xml的内容：

 1 <launch>
  2 
  3   <arg name="x" default="5.0" />
  4   <arg name="y" default="1.0" />
  5   <arg name="z" default="0.0" />
  6   <arg name="yaw" default="3.1415926" />
  7 
  8   
  9   
 10 
 11 
 12   <param name="robot_description_obj" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" />
 13   
 14   <param name="robot_description" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot_dae.xacro" />
 15 
 16   
 17   <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
 18         args="-urdf -x $(arg x) -y $(arg y) -z $(arg z) -Y $(arg yaw) -model xbot-u -param robot_description_obj"/>
 19 
 20   
 23 
 24   
 25   <rosparam file="$(find robot_sim_demo)/param/xbot-u_control.yaml" command="load"/>
 26 
 27   
 28   <node name="spawner" pkg="controller_manager" type="spawner" respawn="false"
 29     output="screen" ns="/xbot" args="joint_state_controller
 30     yaw_platform_position_controller
 31     pitch_platform_position_controller
 32     "/>  
 33 
 34 
 35   
 36   <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" ns="/xbot" respawn="false" output="screen">
 37     <param name="publish_frequency" value="20.0"/>
 38   </node>
 39 
 40   
 41   
 44 
 45   
 46   
 49 
 50   
 51   <node pkg="nodelet" type="nodelet" name="mobile_base_nodelet_manager" args="manager"/>
 52   <node pkg="nodelet" type="nodelet" name="cmd_vel_mux"  args="load yocs_cmd_vel_mux/CmdVelMuxNodelet mobile_base_nodelet_manager">
 53     <param name="yaml_cfg_file" value="$(find robot_sim_demo)/param/mux.yaml"/>
 54     <remap from="cmd_vel_mux/output/cmd_vel" to="cmd_vel"/>
 55   </node>
 56 
 57 
 58 </launch>

KuoYa233, 2020-03-21:
以下是~/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/launch/include/xbot-u.launch.xml的内容：
```
<launch>

  <arg name="x" default="5.0" />
  <arg name="y" default="1.0" />
  <arg name="z" default="0.0" />
  <arg name="yaw" default="3.1415926" />

  <!-- Setup controllers -->
  <!-- rosparam file="$(find fetch_gazebo)/param/freight_controllers.yaml" command="load" / -->

  <param name="robot_description_obj" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" />
  <!-- URDF and TF support -->
  <param name="robot_description" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot_dae.xacro" />

  <!-- Put a robot in gazebo, make it look pretty -->
  <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
        args="-urdf -x $(arg x) -y $(arg y) -z $(arg z) -Y $(arg yaw) -model xbot-u -param robot_description_obj"/>

  <!--node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <param name="use_gui" value="false"/>
  </node-->

  <!--Load the joint controllers to param server-->
  <rosparam file="$(find robot_sim_demo)/param/xbot-u_control.yaml" command="load"/>

  <!--Load controllers-->
  <node name="spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" ns="/xbot" args="joint_state_controller
    yaw_platform_position_controller
    pitch_platform_position_controller
    "/>  <!--mobile_base_controller-->

  <!-- convert joint states to TF transforms for rviz, etc -->
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" ns="/xbot" respawn="false" output="screen">
    <param name="publish_frequency" value="20.0"/>
  </node>

  <!-- Publish base_scan_raw if anything subscribes to it -->
  <!--node name="publish_base_scan_raw" pkg="topic_tools" type="relay" args="base_scan base_scan_raw" >
    <param name="lazy" type="bool" value="True"/>
  </node-->

  <!-- Start a mux between application and teleop, but the switch must be called by service rather than automatically -->
  <!--<node pkg="topic_tools" type="mux" name="cmd_vel_mux" respawn="true" args="/cmd_vel /cmd_vel_mux/input/teleop /cmd_vel_mux/input/navi">
    <remap from="mux" to="cmd_vel_mux"/>
  </node>-->

  <!-- To make the interface of simulation identical to real XBot -->
  <node pkg="nodelet" type="nodelet" name="mobile_base_nodelet_manager" args="manager"/>
  <node pkg="nodelet" type="nodelet" name="cmd_vel_mux"  args="load yocs_cmd_vel_mux/CmdVelMuxNodelet mobile_base_nodelet_manager">
    <param name="yaml_cfg_file" value="$(find robot_sim_demo)/param/mux.yaml"/>
    <remap from="cmd_vel_mux/output/cmd_vel" to="cmd_vel"/>
  </node>
</launch>
```

Twilight89, 2021-01-17:
老哥解决了吗，我也有一样的问题了...