DroidAITech/ROS-Academy-for-Beginners #83: 在我的机器medloic版本下运行教学包里rosbot_sim_demo程序时，终端回溯urdf和xbot-u.launch.xml的异常。老师能提示一下图片command="   "在这种情况下怎样改吗？

KuoYa233, 2020-04-14:
**终端打印信息如下：**
```shell
 roslaunch robot_sim_demo robot_spawn.launch
... logging to /home/hairuo/.ros/log/3bf4a3de-7dfc-11ea-8047-8c16458681e1/roslaunch-hairuo-Lenovo-Legion-Y7000-3020.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

substitution args not supported:  No module named 'rospkg'
when processing file: /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/urdf/robot.xacro
RLException: while processing /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/launch/include/xbot-u.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description_obj]: command [/opt/ros/melodic/share/xacro/xacro.py /home/hairuo/catkin_ws/src/ROS-Academy-for-Beginners-melodic/robot_sim_demo/urdf/robot.xacro] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" name="robot_description_obj"/>
The traceback for the exception was written to the log file

```

**以下是urdf和launch文件截图：**
[image]
[image]

wocut, 2020-04-14:
似乎要下一个melodic版本的教学包，我在gitee 上面找到一个
https://gitee.com/victordoom/ROS-Academy-for-Beginners-melodic
但我找不到melodic 相关的教程，可能是跟原来的教程是一样的吧！

KuoYa233, 2020-04-14:
> 似乎要下一个melodic版本的教学包，我在gitee 上面找到一个
> https://gitee.com/victordoom/ROS-Academy-for-Beginners-melodic
> 但我找不到melodic 相关的教程，可能是跟原来的教程是一样的吧！

谢谢你的建议，不过我运行的就是这个教程的melodic版本程序。
[melodic 相关的教学代码包点击这里](https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic)

littlestarwxy, 2020-05-13:
同样的问题，想问题主解决了吗

KuoYa233, 2020-05-13:
> 同样的问题，想问题主解决了吗

没有呢，太菜，还不会改。这个老师感觉半年多没上来解答问题了，哭～

anchuanxu, 2020-05-27:
> > 同样的问题，想问题主解决了吗
> 
> 没有呢，太菜，还不会改。这个老师感觉半年多没上来解答问题了，哭～

柴老师很久没来了，但是这个包还是一直有人在维护回答问题的23333

anchuanxu, 2020-05-27:
有可能是rospkg的问题，你试试

`sudo apt-get install python-rospkg`

JoeyforJoy, 2020-09-05:
我也遇到了这个问题，后来参考ROS官方论坛上的一个[帖子](https://answers.ros.org/question/331455/xacro-substitution-args-not-supported-no-module-named-rospkg/)解决了。
可能是之前装了Anaconda的原因，一些Python相关的环境遍历被改变了。运行下面的命令重装下rospkg就可以了。
```
conda install -c conda-forge rospkg
```

KuoYa233, 2020-09-12:
> 有可能是rospkg的问题，你试试
> 
> `sudo apt-get install python-rospkg`

你好，你说可能是rospkg问题。我重新安装了一次rospkg，重启后运行教学包代码还是报一开始的错。你还知道别的办法怎样解决这个问题吗？
[image]

KuoYa233, 2020-09-12:
> 我也遇到了这个问题，后来参考ROS官方论坛上的一个[帖子](https://answers.ros.org/question/331455/xacro-substitution-args-not-supported-no-module-named-rospkg/)解决了。
> 可能是之前装了Anaconda的原因，一些Python相关的环境变量被改变了。运行下面的命令重装下rospkg就可以了。
> 
> ```
> conda install -c conda-forge rospkg
> ```

我这个好像安装失败了，发生了意料之外的错误，不太行
[image]

KuoYa233, 2020-09-12:
> 有可能是rospkg的问题，你试试
> 
> `sudo apt-get install python-rospkg`

还有这个问题，说那个包不是launch，可慕课里头好像已经做好launch了
[image]

COMING-X, 2021-12-28:
command [/opt/ros/melodic/share/xacro/xacro.py 的.py 这里改成 xacro -i 就可以了

JoeyforJoy, 2021-12-28:
您好，您的邮件已经收到，我会尽快给您回复。

st929, 2024-12-24:
请问您现在解决了么