DroidAITech/ROS-Academy-for-Beginners #87: service problem

mr7Jacky, 2020-05-28:
... logging to /home/jacky/.ros/log/5bbbae6e-a08a-11ea-aefb-e86f38d11c35/roslaunch-jacky-lin-ubuntu-29241.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:36639/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    client (my_service/service_client.py)
    server (my_service/service_server.py)

ROS_MASTER_URI=http://127.0.0.1:11311

process[server-1]: started with pid [29257]
process[client-2]: started with pid [29258]
Traceback (most recent call last):
  File "/home/jacky/catkin_ws/src/my_service/scripts/service_server.py", line 4, in <module>
    from my_service.srv import my_srv, my_srvResponse
ImportError: No module named my_service.srv
Traceback (most recent call last):
  File "/home/jacky/catkin_ws/src/my_service/scripts/service_client.py", line 4, in <module>
    from my_service.srv import my_srv, my_srvRequest
ImportError: No module named my_service.srv
[server-1] process has died [pid 29257, exit code 1, cmd /home/jacky/catkin_ws/src/my_service/scripts/service_server.py __name:=server __log:=/home/jacky/.ros/log/5bbbae6e-a08a-11ea-aefb-e86f38d11c35/server-1.log].
log file: /home/jacky/.ros/log/5bbbae6e-a08a-11ea-aefb-e86f38d11c35/server-1*.log
[client-2] process has died [pid 29258, exit code 1, cmd /home/jacky/catkin_ws/src/my_service/scripts/service_client.py __name:=client __log:=/home/jacky/.ros/log/5bbbae6e-a08a-11ea-aefb-e86f38d11c35/client-2.log].
log file: /home/jacky/.ros/log/5bbbae6e-a08a-11ea-aefb-e86f38d11c35/client-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
jacky

mr7Jacky, 2020-05-28:
@anchuanxu

anchuanxu, 2020-05-28:
根据log显示,问题出现在

```
File "/home/jacky/catkin_ws/src/my_service/scripts/service_server.py", line 4, in
from my_service.srv import my_srv, my_srvResponse
ImportError: No module named my_service.srv
```

导包语句换成`from my_service.srv import *` 试试,如果还不行,检查自己定义的srv是不是有问题;

### ps:一个小建议

- 学习下markdown语法,这在排版格式上会有帮助,利用好ROS开发社区,上面有很多过去踩过的坑和解决办法,可以省力不少,另外注意提问的完整性,方便答者更好的了解你的问题,从而帮助到你~

mr7Jacky, 2020-05-28:
> 根据log显示,问题出现在
>
> ```
> File "/home/jacky/catkin_ws/src/my_service/scripts/service_server.py", line 4, in
> from my_service.srv import my_srv, my_srvResponse
> ImportError: No module named my_service.srv [...]

好的。我后来改了CMakeList好像就行了。应该不是import的问题

mr7Jacky, 2020-05-28:
@anchuanxu 谢谢

closed 2020-05-28 by mr7Jacky