DroidAITech/ROS-Academy-for-Beginners #92: 运行robot_sim_demo示例时出错

luoxinkaikevin, 2020-06-25:
process[gazebo-1]: started with pid [48854]
process[gazebo_gui-2]: started with pid [48859]
process[urdf_spawner-3]: started with pid [48861]
process[xbot/spawner-4]: started with pid [48865]
process[xbot/robot_state_publisher-5]: started with pid [48866]
process[mobile_base_nodelet_manager-6]: started with pid [48867]
process[cmd_vel_mux-7]: started with pid [48868]
[FATAL] [1593074203.813290445]: ASSERTION FAILED
        file = /home/lxk/catkin_ws/src/ROS-Academy-for-Beginners-melodic/ros_comm-noetic-devel/clients/roscpp
/include/ros/timer_manager.h
        line = 242
        cond = false
        message =
[FATAL] [1593074203.813364622]: ros::TimerManager was instantiated by package roscpp, but neither BOOST_THREA
D_HAS_CONDATTR_SET_CLOCK_MONOTONIC nor BOOST_THREAD_INTERNAL_CLOCK_IS_MONO is defined! Be aware that timers m
ight misbehave when system time jumps, e.g. due to network time corrections.
[xbot/robot_state_publisher-5] process has died [pid 48866, exit code -5, cmd /opt/ros/melodic/lib/robot_stat
e_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/lxk/.ros/log/afa1bfa2-b6a4-11ea-
be1b-000c29aae54c/xbot-robot_state_publisher-5.log].
log file: /home/lxk/.ros/log/afa1bfa2-b6a4-11ea-be1b-000c29aae54c/xbot-robot_state_publisher-5*.log
[mobile_base_nodelet_manager-6] process has died [pid 48867, exit code -5, cmd /opt/ros/melodic/lib/nodelet/n
odelet manager __name:=mobile_base_nodelet_manager __log:=/home/lxk/.ros/log/afa1bfa2-b6a4-11ea-be1b-000c29aa
e54c/mobile_base_nodelet_manager-6.log].
log file: /home/lxk/.ros/log/afa1bfa2-b6a4-11ea-be1b-000c29aae54c/mobile_base_nodelet_manager-6*.log
[INFO] [1593074204.268881, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_control
ler
[cmd_vel_mux-7] process has died [pid 48868, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet load yocs
_cmd_vel_mux/CmdVelMuxNodelet mobile_base_nodelet_manager cmd_vel_mux/output/cmd_vel:=cmd_vel __name:=cmd_vel
_mux __log:=/home/lxk/.ros/log/afa1bfa2-b6a4-11ea-be1b-000c29aae54c/cmd_vel_mux-7.log].
log file: /home/lxk/.ros/log/afa1bfa2-b6a4-11ea-be1b-000c29aae54c/cmd_vel_mux-7*.log

huochen1, 2021-03-12:
老哥，解决了嘛

luoxinkaikevin, 2021-03-15:
> 老哥，解决了嘛
一直没解决。。。