DroidAITech/ROS-Academy-for-Beginners #103: 仿真启动失败

YunxiangLuo, 2020-08-17:
xbot@xbot-ThinkPad-T470p:~$ roslaunch robot_sim_demo robot_spawn.launch 
... logging to /home/xbot/.ros/log/6087e926-e063-11ea-8d30-74da38f391f1/roslaunch-xbot-ThinkPad-T470p-22869.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
xacro.py is deprecated; please use xacro instead
started roslaunch server http://127.0.0.1:44169/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/xbot/catkin...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /robot_description_obj: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.7
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)

auto-starting new master
process[master]: started with pid [22885]
ROS_MASTER_URI=http://127.0.0.1:11311

setting /run_id to 6087e926-e063-11ea-8d30-74da38f391f1
process[rosout-1]: started with pid [22896]
started core service [/rosout]
process[gazebo-2]: started with pid [22902]
process[gazebo_gui-3]: started with pid [22908]
process[urdf_spawner-4]: started with pid [22913]
process[xbot/spawner-5]: started with pid [22914]
process[xbot/robot_state_publisher-6]: started with pid [22915]
process[mobile_base_nodelet_manager-7]: started with pid [22916]
process[cmd_vel_mux-8]: started with pid [22917]
gzserver: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
gzclient: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[gazebo-2] process has died [pid 22902, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/xbot/catkin_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/worlds/ROS-Academy.world __name:=gazebo __log:=/home/xbot/.ros/log/6087e926-e063-11ea-8d30-74da38f391f1/gazebo-2.log].
log file: /home/xbot/.ros/log/6087e926-e063-11ea-8d30-74da38f391f1/gazebo-2*.log
[INFO] [1597652802.997162, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[gazebo_gui-3] process has died [pid 22908, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/xbot/.ros/log/6087e926-e063-11ea-8d30-74da38f391f1/gazebo_gui-3.log].
log file: /home/xbot/.ros/log/6087e926-e063-11ea-8d30-74da38f391f1/gazebo_gui-3*.log
[INFO] [1597652803.412223, 0.000000]: Loading model XML from ros parameter robot_description_obj
[INFO] [1597652803.439486, 0.000000]: Waiting for service /gazebo/spawn_urdf_model

anchuanxu, 2020-08-17:
执行：`sudo apt-get update && upgrade`

anchuanxu, 2020-09-16:
该问题已解决

closed 2020-09-16 by anchuanxu