DroidAITech/ROS-Academy-for-Beginners #109: ubuntu20 ROS neotic 能用这个文件吗

lwt-414, 2020-09-16:
ubuntu20 ROS neotic 能用这个文件吗
安装所需依赖要怎么改命令行

anchuanxu, 2020-09-16:
你好 没有试过哦，可以尝试下，安装依赖将对应的melodic替换成neotic就行了

lwt-414, 2020-09-17:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
slam_sim_demo: Cannot locate rosdep definition for [gmapping]
navigation_sim_demo: Cannot locate rosdep definition for [gmapping]
robot_sim_demo: Cannot locate rosdep definition for [yocs_cmd_vel_mux]
显示这个问题要怎么办呢 在网上找了好多方法都不行

hoaklee, 2020-10-19:
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> slam_sim_demo: Cannot locate rosdep definition for [gmapping]
> navigation_sim_demo: Cannot [...]

https://github.com/DroidAITech/ROS-Academy-for-Beginners/issues/13#issuecomment-511379240

hoaklee, 2020-10-19:
请问在20.04上还能使用吗。libqt4-dev被版本移除了，安装时候提示缺失软件包

> 你好 没有试过哦，可以尝试下，安装依赖将对应的melodic替换成neotic就行了

hu-jinfei, 2020-10-21:
> libqt4-dev被版本移除了，安装时候提示缺失软件包
> 
> >你好，请问你解决了吗，缺少qt4依赖项

lwt-414, 2020-10-21:
没办法 我换ubuntu18了

kaigemima, 2020-10-31:
希望能出一个针对20.04的教程包，现在一些东西已经开始慢慢地不再支持16.04了

olivener, 2020-12-23:
ubuntu20.04的依赖包更改版本也不行，希望能够出一款针对20.04的教程包

Protect-hair, 2021-05-12:
有没有在Ubuntu20.04上解决了的？

kaigemima, 2021-05-28:
没解决，后来没搞那个了

17775208950, 2021-07-09:
安装和编译会有一些小问题，但是百度基本都能解决。但是加载模型的时候会有一些警告，并且在zagebo里看不到模型，只有几个小组件。使用collision view可以看到模型结构，但是没有纹理和贴图

zhu608, 2022-03-18:
> 安装和编译会有一些小问题，但是百度基本都能解决。但是加载模型的时候会有一些警告，并且在zagebo里看不到模型，只有几个小组件。使用collision view可以看到模型结构，但是没有纹理和贴图

我是18.04也是同样情况，请问怎么解决

FanShupei, 2022-03-28:
我用以下命令解决了
```
rosdep install --from-paths src --ignore-src --rosdistro=noetic -r -y
```

zhu608, 2022-03-29:
> 我用以下命令解决了
> 
> ```
> rosdep install --from-paths src --ignore-src --rosdistro=noetic -r -y
> ```

非常感谢，18.04把noetic改成melodic就可以了