DroidAITech/ROS-Academy-for-Beginners #113: launch文件启动时报错求解

lightersp, 2020-09-23:
RLException: while processing /home/yanzhenwang/桌面/catkin_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/launch/include/xbot-u.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: no such command [['/opt/ros/noetic/share/xacro/xacro.py', '/home/yanzhenwang/桌面/catkin_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/urdf/robot.xacro']]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro"/>
The traceback for the exception was written to the log file
出现这个怎么办

lightersp, 2020-09-25:
已经解决了 打开相应XML文件把xacro.py的.py去掉就好了

lightersp, 2020-09-25:
另外这个demo模型在除了knetic其他版本都无法打开 求更新

CheungChris, 2021-01-13:
已经更新了呀

zhu608, 2022-03-18:
> 已经更新了呀

你好，18.04只有地板没有机器人怎么弄出来