DroidAITech/ROS-Academy-for-Beginners #115: roslaunch robot_sim_demo robot_spawn.launch 错误

ccwenyi, 2020-10-16:
RLException: while processing /home/bajie/droidAB_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/launch/include/xbot-u.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/share/xacro/xacro.py', '/home/bajie/droidAB_ws/src/ROS-Academy-for-Beginners/robot_sim_demo/urdf/robot.xacro']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py $(find robot_sim_demo)/urdf/robot.xacro" name="robot_description"/>
The traceback for the exception was written to the log file

ccwenyi, 2020-10-16:
以解决

closed 2020-10-16 by ccwenyi

YTianZzz, 2021-02-09:
如何解决的能说一下吗

Ding-Guo, 2022-02-24:
您好！我也遇到这个问题了，请问是怎么解决的

ccwenyi, 2022-02-24:
好像是Gazebo版本太低，记不太清了

Ding-Guo, 2022-02-24:
感谢,已经解决了