DroidAITech/ROS-Academy-for-Beginners #133: ubuntu18.04 gazebo仿真的时候所有都正常就是没有小车模型

weiwei366, 2021-04-09:


weiwei366, 2021-04-12:
> _No description provided._

... logging to /home/yjw/.ros/log/8370427e-9b36-11eb-b1d3-08d23e95ee0d/roslaunch-yjw-G3-10867.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
xacro.py is deprecated; please use xacro instead
started roslaunch server http://yjw-G3:33655/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/yjw/catkin_...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /robot_description_obj: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [10888]
process[gazebo_gui-2]: started with pid [10893]
process[urdf_spawner-3]: started with pid [10898]
process[xbot/spawner-4]: started with pid [10899]
process[xbot/robot_state_publisher-5]: started with pid [10900]
process[mobile_base_nodelet_manager-6]: started with pid [10901]
process[cmd_vel_mux-7]: started with pid [10902]
[INFO] [1618195857.834582, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1618195857.955533553]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1618195857.956724303]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/gazebo_ros/spawn_model", line 34, in <module>
    from tf.transformations import quaternion_from_euler
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 39, in <module>
    from .buffer_interface import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/buffer_interface.py", line 32, in <module>
    import roslib; roslib.load_manifest('tf2_ros')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 64, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 97, in _generate_python_path
    m = rospack.get_manifest(pkg)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 171, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 215, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 414, in parse_manifest_file
    _static_rosdep_view = init_rospack_interface()
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 134, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 603, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 561, in load_cached_sources_list
    with open(cache_index, 'r') as f:
IOError: [Errno 13] Permission denied: '/home/yjw/.ros/rosdep/sources.cache/index'
[ INFO] [1618195857.991850287]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1618195857.992884697]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[urdf_spawner-3] process has died [pid 10898, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -urdf -x 5.0 -y 1.0 -z 0.0 -Y -3.1 -model xbot-u -param robot_description_obj __name:=urdf_spawner __log:=/home/yjw/.ros/log/8370427e-9b36-11eb-b1d3-08d23e95ee0d/urdf_spawner-3.log].
log file: /home/yjw/.ros/log/8370427e-9b36-11eb-b1d3-08d23e95ee0d/urdf_spawner-3*.log
[ INFO] [1618195858.791261869]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1618195858.804069425]: Physics dynamic reconfigure ready.
[WARN] [1618195887.913130, 28.910000]: Controller Spawner couldn't find the expected controller_manager ROS interface.

xiosi, 2022-05-04:
我也遇到了这个问题，我没弄明白错误发生的原因！但通过改相应xacro中的mesh文件后缀为obj后成功了。