DroidAITech/ROS-Academy-for-Beginners #156: 运行仿真时存在bug:"gazebo_ros_control plugin: GazeboRosControlPlugin missing <legacyModeNS> DefaultRobotHWSim"

mingdonghu, 2023-04-05:
- 解决办法：
在gazebo_ros_control包的调用地方配置上<legacyModeNS>true</legacyModeNS>，如下所示：
```xml
<gazebo>
    <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
      <robotNamespace>/xbot</robotNamespace>
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
      <legacyModeNS>true</legacyModeNS>
    </plugin>
  </gazebo>
```

Ycysara, 2024-05-27:
请问要改的这个文件名是什么

mingdonghu, 2024-05-27:
具体不记得了，你可以用vscode打开它这个工程，在搜索里面输入这些文段的关键字

|
胡明栋David Hu
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年5月27日 14:57 |
| 收件人 | ***@***.***> |
| 抄送人 | David ***@***.***>,
***@***.***> |
| 主题 | Re: [DroidAITech/ROS-Academy-for-Beginners] 运行仿真时存在bug:"gazebo_ros_control plugin: GazeboRosControlPlugin missing <legacyModeNS> DefaultRobotHWSim" (Issue #156) |

请问要改的这个文件名是什么